srobo/docs #14: Add simulator zip files

richardbarlow, 2016-09-22:
There are links on https://studentrobotics.org/docs/programming/simulator.html to three zip files for the simulator. I don't really know anything about the simulator, but we've already had a team asking about it. Since some teams clearly use it, I don't see why we shouldn't keep hosting it as-is until further notice.

richardbarlow, 2016-09-22:
I've just looked into this a little bit. Oh the horror. srweb.git contains three zip files, each of which has a clone of simulator.git and a special txt file that is, from what I can tell, not stored anywhere else in the world. I feel that this needs more work than just copying across some files.

I'll work on this when I'm back off holiday. I plan on moving the three magic txt files into the simulator.git repo. Hosting the repo on github and changing the links on the website to github zip links.

richardbarlow, 2016-09-22:
(That is assuming there is no other magicness in the individual zips that I am yet to uncover)

PeterJCLaw, 2016-09-22:
Note that the simulator will need updating for this year's game, once that is decided. For reference, here's the ticket for last year's update: https://www.studentrobotics.org/trac/ticket/3109

PeterJCLaw, 2016-09-22:
Note also that alongside the archives is [a script](https://www.studentrobotics.org/cgit/srweb.git/tree/resources/simulator/update.sh) which was created for updating them and added as part of last year's update.

richardbarlow, 2016-09-28:
Since the existing zips work (I assume) and Peter has already written a script to update them, I am going to just commit them to the website repo for now and work to improve the deployment of the simulator can happen in the future.

closed 2016-10-10 by richardbarlow