srobo/docs #97: Document 32 buzz buffer limit

trickeydan, 2019-03-10:
Competitors are getting USB errors and couldn't understand why. This is due to the ring buffer on the power board overflowing.

PeterJCLaw, 2021-11-20:
Does this limit still apply with the new generation of kit software?

trickeydan, 2021-11-20:
> Does this limit still apply with the new generation of kit software?

The limit is caused by the size of a ring buffer in the power board firmware. The board firmwares are scheduled for a review and improvements, but we are unlikely to fix this issue until SR2023.

PeterJCLaw, 2021-11-20:
I guess I was wondering whether the new Python API handled this in a way that resulted in nicer errors or something, which might be doable in the meanwhile (though is obviously fairly low priority/may not be worth it).

trickeydan, 2021-11-20:
> I guess I was wondering whether the new Python API handled this in a way that resulted in nicer errors or something, which might be doable in the meanwhile (though is obviously fairly low priority/may not be worth it).

Oh we have nicer errors for it, yes: https://github.com/j5api/j5/blob/main/j5/backends/hardware/sr/v4/power_board.py#L196-L200

WillB97, 2021-11-20:
Out of interest, what state is the power board left in when it crashes. Namely do the outputs turn off.

(I'm aware that this discussion should probably be on srobo/power-v4-fw#3 but more people in the know seem to be here)

trickeydan, 2021-11-20:
> Out of interest, what state is the power board left in when it crashes? Namely do the outputs turn off?

I'm afraid I don't know, but we should definitely test this.

WillB97, 2022-10-02:
The new firmware has no ring buffer so this is no longer relevant.

trickeydan, 2022-10-02:
This has been fixed by new power board firmware

closed 2022-10-02 by trickeydan