srobo/docs #350: Update Interactive Troubleshooter

PeterJCLaw, 2022-10-02:


PeterJCLaw, 2022-10-02:
I'm not actually sure what needs updating here. I had a run through of the troubleshooter earlier today and didn't spot anything major kit-wise. (I did find an issue where the links didn't work though; https://github.com/srobo/docs/pull/351 fixes that)

#311 is already recording the WiFi side of things, however if we are shipping WiFi and there is a manual interface on the brain board then that doesn't actually need to change (the WiFi docs will need updating, but that's separate).

@trickeydan or @raccube is there more detail on this task that you're aware of?

raccube, 2022-10-13:
Considering we now have the KCH, this could maybe do with a rewrite.

PeterJCLaw, 2022-10-13:
I'm not sure I follow? Why does the KCH mean we need a rewrite of the Interactive Troubleshooter? Most of the content there is about boards which are still part of the kit. There's likely parts which need updating, but I imagine that most of it is fine.

sedders123, 2022-10-21:
I've opened #384 to help towards this

sedders123, 2022-10-21:
We think this is in a good enough state for Kickstart

closed 2022-10-21 by sedders123