srobo/docs #451: Add documentation on simulator "competition mode"

RealOrangeOne, 2023-02-24:
We mention in passing to competitors about "competition mode", and how some things are a little different, but we don't document _how_ to use competition mode, and canonically what the differences are.

See https://github.com/srobo/competition-simulator#competition-mode

PeterJCLaw, 2023-10-13:
Note that the README.md you've linked to here is included in the archive which distributes the simulation, so this kinda already is documented. I suspect the issue is that many competitors won't read that document, so there probably is value in repeating the instruction in the main docs.

PeterJCLaw, 2023-10-21:
I had a quick look at doing this this evening and it's not totally simple to do this as a competitor given the need to put a file in the right place and install all the libraries. You don't need to provide a match file nor robot code for all zones though.