srobo/docs #488: Docker for rake validate

8BitJosh, 2023-08-28:
Add some sort of Docker based tool for validating (check links/spelling) the docs locally.

PeterJCLaw, 2023-08-28:
Could you expand on why Ruby/Rake are a pain on Windows? Is this a general issue in the ecosystem or something specific to how we're set up? Would it be useful for example for us to have more info on how to set them up and/or guidance on e.g: using them under WSL (assuming that that might be better than under plain Windows)?

There already is a Docker setup I think in the docs repo, though I don't use it. I had thought it already supported running the validate stuff. Personally I find its presence is a negative as it makes the CI slower. I'm on Linux (Ubuntu) though, and am able to run everything locally. It also therefore wouldn't occur to me to update it if things changed, which means it's liable to get out of sync.

For spell checking if you're using VSCode I'd encourage installing the Code Spell Checker extension. That should run the same spell checker in your editor as we run in CI. This is the main reason that we use the checker package that we do. I see however that that wasn't being suggested to users of the repo though, I've fixed that now (copied over from the wesite repo).
I don't know if it's supported in other editors, though it feels like it easily could be.

PeterJCLaw, 2024-03-02:
https://github.com/srobo/docs/pull/585 adds more information on how to run the checks locally.

WillB97, 2024-03-03:
That doesn't solve this problem though. This is about being able to use the validate tools with docker, not running them locally.

PeterJCLaw, 2024-03-03:
#585 attempts to address the reason for this request -- that setting these up is hard, which I've assumed is due to lack of information about what's needed. I've asked for further information about why it's hard but haven't had a response -- hence the guesswork.